thonny/thonny #3399: Pygame module not in pypi package in install package option

spoonsandhangers, 2024-12-14:
On our school network we rely on installing packages using the tools manage packages/plugins options. Students do not have access to a terminal as it is blocked by our network.

Students can no longer Install pygame in this way. The search results are blank.

Any help would be appreciated but we cannot use pip as no command line access.

Thank you

jumping-albatross, 2024-12-15:
Here's a "me too". I've now seen this problem replicated on Windows 10, macOS and Linux Mint. Instead of pygame it's pygame zero that I use :)

There are some ways around the problem but they tend to require admin access, or, perhaps, at the very least, access to the AppData folder on Windows (both of which are blocked by group policy on school computers). On personal devices or with admin access it's possible to use the 'whl' file to install. Or, you can install from a source download.

My guess is that PyPI broke something on the server end :(

aivarannamaa, 2024-12-15:
Please see #3401

closed 2024-12-15 by aivarannamaa

aivarannamaa, 2024-12-16:
Please try https://github.com/thonny/thonny/releases/tag/v4.1.7