thonny/thonny #3405: No longer access to packages

lulu-vivi, 2024-12-17:
Since recently, it's become impossible to install or even search for new packages. I'm new to Thonny, is there a fee to pay after a while ? So far there was no problem to search and install new packages...

aivarannamaa, 2024-12-17:
Please try version 4.1.7

mblasee, 2024-12-17:
Not working for me either, using 4.1.7 on Windows 11

aivarannamaa, 2024-12-17:
@mblasee, please re-check your version. If you really have 4.1.7, then please list the steps for reproducing the problem - including where you clicked, what you typed and what happened.

mblasee, 2024-12-17:
https://github.com/user-attachments/assets/02bf8325-d583-451b-a48c-57f72e9a3664

I recorded what I was doing but here are the steps as well

I typed help() to the repl to show that its not frozen and Thonny installation/esp firmware is fine.
I opened Help/About Thonny to verify my version, its 4.1.7
I opened Tools/Manage Packages typed aioble and pressed enter. Then nothing happens no matter how long I am waiting.

mblasee, 2024-12-17:
I was poking around a bit and seems like if I use "Local Python 3  ° Thonny's Python" mode it's working.
I was under the impression that this package install should work in esp32 mode as well, am I mistaken?

SolarBites, 2024-12-18:
@mblasee is it still working? mine wasnt working when i had the 4.1.6 version, but i downloaded the 4.1.7 and everything is fine. I can download packages which i couldnt in 4.1.6

aivarannamaa, 2024-12-19:
@mblasee, I see that the search button in your case is disabled. This may mean that Thonny is not ready with listing your installed packages.

Does the button change when you wait for 10 seconds after opening the package dialog?

If not, then please check which files you have in your `sys.path` directories. Here's how I check this:

```
>>> import sys
>>> sys.path
['', '/', '/lib']
>>> import os
>>> os.listdir("/lib")
[]
>>> os.listdir("/")
['sd', 'settings.toml', 'code.py', 'lib', 'boot_out.txt']
```

mblasee, 2024-12-19:
I don't know what happened since I didn't change any config or installed anything but it works now. Both in Local Python 3 ° Thonny's Python or MicroPython (ESP32) mode. 

Thanks for everyone's attention, I am fine with closing this issue.

aivarannamaa, 2024-12-20:
Great!

closed 2024-12-20 by aivarannamaa