thonny/thonny #3406: Update python 3.10 version to 3.10.16

crispinboylan-int, 2024-12-17:
Currently thonny windows exe and portable contains 3.10.11 which has many security issues.

This is dinged by Microsoft Defender as containing crit vulnerabilities.  Can the package be updated to latest 3.10.16?

aivarannamaa, 2025-01-15:
Thonny 5.0 will come bundled with Python 3.13

closed 2025-01-15 by aivarannamaa

crispinboylan-int, 2025-02-05:
my point is mainly not the major release, but the patch release is out of date.  thonny 5 may well come with 3.13 but e.g. if it stays on 3.13.0 for the life of thonny 5 that leads to using a python with vulnerabilities.  the download for whatever version is the latest should contain a python with the latest point release (for thonny 4 thats 3.10.16), and be regularly updated to newer point releases, especially when those contain critical or high severity vulnerabilities.

aivarannamaa, 2025-02-08:
I would give such concerns a higher priority if Thonny was meant to be used in a server environment, but I fail to see the risk in using Python version with vulnerabilities in desktop environment.

crispinboylan-int, 2025-02-10:
there's several issues here:
* vulns aren't just exploited on server environments, the same reason windows updates exist for non server environments
* people do use thonny in environments where machines are scanned for vulnerabilities and this may be problematic
* not just vulnerabilities but bugs are fix in the patch releases
* i dont feel like its that much of a hassle to do this, even if it were just including the latest python point release at the same time as a thonny release rather than specifically making efforts to update the python out of band

aivarannamaa, 2025-02-10:
I've updated bundled Python minor versions for each new major Thonny version. For other Thonny updates I have bundled the latest patch version of the same minor version, which have official binary builds. Unfortunately Python maintainers don't publish builds for all patch versions. I may switch to unofficial builds in the future, but this may require non-trivial changes in installers, especially on macOS.