thonny/thonny #3416: [Linux] Thonny won't let me save files anymore

am-subscr, 2024-12-23:
I've just noticed that I can't save files anymore with Thonny on my computer. It (eg "save","save as", save on exit,...) just does nothing...

Could it be linked to the fact I recently upgraded to Ubuntu 24.04.01 LTS? Also, I've noticed while trying to solve the problem that I actually have 2 (conflicting?) different versions installed: 4.0.1, with Synaptic (I tried to uninstall and reinstall - didn't work), and 3.0.13, installed through a bash script - I have no clue how to uninstall this one, it doesn't even appear on Synaptic... By the way, I do experience the same problem on both versions.

Any help will be greatly appreciated!

aivarannamaa, 2024-12-23:
Thonny by default uses `zenity` for file dialogs and it sometimes creates the dialogs behind Thonny's main window. In newer Thonny versions you can switch from zenity to Tk dialogs in Tools => Options => General.

For uninstalling the one installed with a Bash script -- there should be an executable named `uninstall` in `~/apps/thonny/bin`.

am-subscr, 2024-12-24:
Thanks a lot for your help, aivarannamaa! Both workarounds worked perfectly!!!
And thanks for backing me, speedy-10! It helped knowing I was not alone to face this problem. As far as I'm concern this issue is solved!
Best regards!

closed 2025-01-15 by aivarannamaa