thonny/thonny #3424: Remove "Support Ukraine" Button from Thonny

github12101, 2025-01-01:
**Issue:**
The "Support Ukraine" button in Thonny links to politically charged content. As a user from Switzerland, where neutrality is paramount, I believe software should remain neutral and not promote specific political agendas. One cannot solely blame one side or praise the other without considering the complexities of the situation. Embedding such content risks alienating users and detracts from the software's technical focus.

**Request:**
Please remove the "Support Ukraine" button to maintain neutrality and avoid introducing non-technical content into the project. Until this is done, I refuse to use Thonny.

aivarannamaa, 2025-01-02:
I refuse to remain neutral in this issue and I can live with you not using Thonny.

closed 2025-01-02 by aivarannamaa

github12101, 2025-01-07:
> I refuse to remain neutral in this issue and I can live with you not using Thonny.

Great. And I can live without using garbage software like Thonny. Wiped clean out of my harddrive. Have fun without me :)