thonny/thonny #3425: How to programmatically detect running in Thonny (shell) environment

commongeek, 2025-01-02:
I need to be able to do something like this, is it possible?

```
if in_thonny():
    print('Cannot use IPython in Thonny shell')
else:
    from IPython import embed
    embed()
```

But how to write the in_thonny() function?

BertLindeman, 2025-01-02:
Depends on the micropython version.
On my pico2w I use
'MicroPython-1.25.0-preview-arm--with-newlib3.3.0'
And here os has no environ.

In the REPL ```dir()``` shows:
```
['machine', '__thonny_result', '__name__', '__thonny_names', 'rp2', '__thonny_helper', '__thonny_name']
```

```__thonny_names``` shows the filenames found on the MCU.

Maybe you can use the existence of some of the __thonny* variables?

commongeek, 2025-01-02:
I use plain Python interpreter so [aivarannamaa](https://github.com/aivarannamaa) method works for me. In the REPL I don't have `__thonny_*` variables so the question of whether there is a universal method remains open.  ChatGPT proposes:

```
importlib.util.find_spec("thonny") is not None
```

BertLindeman, 2025-01-02:
> I use plain Python interpreter so [aivarannamaa](https://github.com/aivarannamaa) method works for me. In the REPL I don't have `__thonny_*` variables so the question of whether there is a universal method remains open. ChatGPT proposes:
> 
> ```
> importlib.util.find_spec("thonny") is not None
> ```

Probably not universal 😄 
My beta version does not have importlib so chatGPT fails for me.
In fact none of the suggestions I  got from chatGPT does not work for me. 
Updated to Thonny 4.1.7. 

No help found, sorry.

Bert

aivarannamaa, 2025-01-02:
ChatGPT's suggestion only works with the Python interpreter bundled with Thonny.

For MicroPython, @BertLindeman's approach is best there is in current Thonny version.
Thonny 5 will offer a bit nicer alternative: #2907

closed 2025-01-02 by aivarannamaa