thonny/thonny #3426: def main():

SantosenMisty, 2025-01-02:
import urllib.request

#hiermee worden de woorden opgehaald en als lijst geretourneerd.
def fetch_words():
    with urllib.request.urlopen('http://sixty-north.com/c/t.txt') as story:
        story_words = []
        for line in story:
            line_words = line.decode('utf-8').split()
            for word in line_words:
                story_words.append(word)
    return story_words

#hiermee wordt een lijst woorden afgedrukt.
def print_words(story_words):
    for word in story_words:
        
def main():
    words = fetch_words()
    prind_words(words)

MartinPacker, 2025-01-02:
You've opened this as an Issue. I wonder why.

BestGiter, 2025-01-12:
oh wait, that person meant print_words(words), not prind_words(words), good day whoever made this issue

closed 2025-01-15 by aivarannamaa