thonny/thonny #3427: [Bug] Can't install Pi Pico FW with Thonny 4.1.7

vguttmann, 2025-01-04:
I was following the Raspberry Pi tutorial on getting started with a Pi Pico, but when getting to install the firmware, I just got the following error: 
```
Traceback (most recent call last):
  File "/home/vguttmann/apps/thonny/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/vguttmann/apps/thonny/lib/python3.10/site-packages/thonny/workbench.py", line 874, in _post_backend_menu
    if uf2_device_is_present_in_bootloader_mode():
  File "/home/vguttmann/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/uf2dialog.py", line 250, in uf2_device_is_present_in_bootloader_mode
    for vol in list_volumes(skip_letters=["A"]):
  File "/home/vguttmann/apps/thonny/lib/python3.10/site-packages/thonny/misc_utils.py", line 112, in list_volumes
    return list_volumes_sync()
  File "/home/vguttmann/apps/thonny/lib/python3.10/site-packages/thonny/udisks.py", line 19, in list_volumes_sync
    return loop.run_until_complete(list_volumes())
  File "/home/vguttmann/apps/thonny/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/vguttmann/apps/thonny/lib/python3.10/site-packages/thonny/udisks.py", line 120, in list_volumes
    mount_point = await interface.call_mount({})
  File "/home/vguttmann/apps/thonny/lib/python3.10/site-packages/dbus_next/aio/proxy_object.py", line 92, in method_fn
    BaseProxyInterface._check_method_return(msg, intr_method.out_signature)
  File "/home/vguttmann/apps/thonny/lib/python3.10/site-packages/dbus_next/proxy_object.py", line 62, in _check_method_return
    raise DBusError._from_message(msg)
dbus_next.errors.DBusError: Error creating directory `/dev/sda1' to be used for mounting /dev/sda1: Not a directory
```

This happened both in the Flatpak and the recommended installer on Ubuntu 22.04

aivarannamaa, 2025-01-15:
Thanks for the report!

@jwillikers, do you have any ideas about this?

jwillikers, 2025-01-29:
> Thanks for the report!
> 
> [@jwillikers](https://github.com/jwillikers), do you have any ideas about this?

I'm not exactly sure. It's strange that it would attempt to create the directory `/dev/sda1` since that's a Linux device node. Since I can't reproduce locally, at least on Fedora, and this issue hasn't had a lot of traffic, I'd guess that this is related to something specific on the system.

@vguttmann Which variant of the Pico does this occur with, the original Pico, Pico 2, Pico W, etc? Also, do you have any bespoke system configuration you can think of that could effect mounting disks or DBus? Also, are you able to manually install the UF2 to the Pico via drag-n-drop?

vguttmann, 2025-01-29:
I'm working with the Pico W here, on Ubuntu 22.04 LTS.
I'm a member of the dialout group, and I'm able to mount and flash new firmware manually - only Thonny fails.
When using VS Code, I can also connect to the Pi Pico and access the MicroPython shell.

jwillikers, 2025-01-29:
> I'm working with the Pico W here, on Ubuntu 22.04 LTS. I'm a member of the dialout group, and I'm able to mount and flash new firmware manually - only Thonny fails. When using VS Code, I can also connect to the Pi Pico and access the MicroPython shell.
> 
> Are there any other tests I can run/debug flags to set?

The output of the `lsblk` command, with the Pico connected in bootloader mode, would be helpful. You might also run something like `ls -l /dev/sda*` to give the permissions for the device node representing the Pico's disk.

I'll try setting up an Ubuntu 22.04 VM to see if I can reproduce in that environment. I imagine I'll figure out some more in-depth commands you can run for me to get some diagnostic output if I still can't reproduce in the VM. I've got a Pico W, which is what I tested with earlier.