thonny/thonny #3429: The program does not stop.

wss6630, 2025-01-05:
When I press the stop button in the IDE, the program does not stop. CTRL+C does not help, and even a hard reset does not resolve the issue. One time I had to format the ESP32 to continue working. I have observed that this problem happens when _thread, timer, or asyncio is used in the code, and boot.py starts main.py. When boot.py does not start anything, there is no problem. This issue occurs on ESP32(+CH340), ESP32 C3(+usb/jtag), and ESP32 S2(+usb).

wss6630, 2025-01-11:
I noticed that this problem is only on dual-core CPU. On single-core systems, the same code is interrupted without problems.

aivarannamaa, 2025-01-15:
May be related to #3324

Possibly also: https://github.com/thonny/thonny/issues?q=is%3Aissue+is%3Aopen+dual+core