thonny/thonny #3431: Request of a new option

NeilL67, 2025-01-07:
Can we please have an option to NOT save the code every time it's run ?
Happy for this to be a checkbox under options or via an environment variable.
The way I have always worked (under many different languages and IDEs) is to save my code at stages where it has been tested, so it's always stable. I'm finding I can change a couple of things, break it, then not be able to revert back to the last working version to work out what I've done.
Many thanks.

NeilL67, 2025-01-19:
Hi, thanks for getting back to me. No, I want to run the version in the window, I just want to control the saving only when I want it to be saved, e.g. at a working checkpoint each time. This means if I break something I can go back to the previous save (just like playing a video game). Many thanks.

aivarannamaa, 2025-01-19:
Can you give an example of another IDE or editor, which allows this?

NeilL67, 2025-01-19:
Notepad, Arduino IDE, Delphi, MS Word, everything I've ever used in fact. They all only save when you tell them to, and will warn if you close without saving the latest changes. I've only ever seen this behaviour in Thonny.
Thanks.

aivarannamaa, 2025-01-20:
What do you expect to happen, when an editor has unsaved edits and you run it?

NeilL67, 2025-02-27:
Apologies, I didn't see you new question.
I expect it to run, but not to save it first. 
If I exit without it being saved I would like it remind me that it has not been saved.
This is the behavior I have personally always seen from every editor and IDE in the last 40 years of working in IT as a developer.

aivarannamaa, 2025-03-01:
> I expect it to run, but not to save it first.

Which version of the code you expect to be run? The one in the editor or the one in the file?

NeilL67, 2025-03-02:
The one in the editor, this is the point I am trying to make. 
While making changes, I want to be able to save only when I am happy that the changes I have just made work, else I can rollback to the previous working version in case it's not obvious what change has broken things. Just the same as saving a checkpoint while playing a game.
Thanks.

aivarannamaa, 2025-03-02:
Well, let's get back to this:

> Can you give an example of another IDE or editor, which allows this?

My idea was check out how running without saving is presented in other editors. You said:

> Notepad, Arduino IDE, Delphi, MS Word, everything I've ever used in fact

This got me confused.

* I'm not aware of any way Notepad or MS Word can **run** something. Can you elaborate?
* It has been long since I last used Delphi, but doesn't it require to save the file(s) before compiling / running?
* I haven't used Arduino IDE. Can you confirm that when I save an editor (i.e. associate it to a file), then make some changes and then upload the sketch, it compiles the version in the editor without updating the file first?

NeilL67, 2025-03-03:
Notepad cannot run anything, you said other editors, notepad is a text editor.
I've not used Delphi for a couple of years as I have now retired, but from memory (have used it for nearly 20 years, many versions), it ran the current state of the IDE, and only saved to disc when you told it to.
The Arduino IDE only saves when told to to PC, it does upload the current session to the board, but it does not overwrite the file stored on the PC (and you can't access the code directly from the Arduino as you can the Pico as it's compiled and not interpreted).
I didn't realise this would cause so much trouble, so please don't worry about it, it was only a suggestion. I'm saving different versions to different filenames to get round the restriction.
Thanks,
Neil

aivarannamaa, 2025-03-21:
I'm closing this issue, as I'm afraid running from the named editor without updating underlying file would make matters too confusing for most users.

Please note that Thonny 5 will have refactored "Replayer", which makes it easy to recover any previous state of the code.

closed 2025-03-21 by aivarannamaa