thonny/thonny #3437: The display area of debug variables is too narrow, and it is not very convenient for user， I need to drag the window every time.  Can thonny increase the default height of the display area of debug variables?

wilikeit, 2025-01-16:
The display area of debug variables is too narrow, and it is not very convenient for user， I need to drag the window every time.  Can thonny increase the default height of the display area of debug variables?
[image]

_Originally posted by @wilikeit in https://github.com/thonny/thonny/discussions/3382_

aivarannamaa, 2025-01-16:
Thanks! Will investigate soon.

chandradharkoneti, 2025-12-12:
Hi @aivarannamaa 
I have this issue too (not just in debugging mode, but also in regular mode when I open the variables viewer).
I would love to be able to move the variables pane to the same 'pane-block' as `assistant` & `object inspector`
Currently, it looks like so:

[image]

When I open the variables pane, it makes the shell window too small.
Is there any way in which I can move the variables pane down, so that it solves 2 issues. 

1. It increases the size of variables pane, so that more variables are visible at once.
2. It does not eat up precious shell (REPL window) space.