thonny/thonny #3441: os.get_terminal_size()

MdV-dev, 2025-01-19:
### In a Windows command line windows :
Microsoft Windows [version 10.0.26100.2894]
(c) Microsoft Corporation. Tous droits réservés.

C:\Users\michel>python
Python 3.13.0 (tags/v3.13.0:60403a5, Oct  7 2024, 09:38:07) [MSC v.1941 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> print(os.get_terminal_size())
os.terminal_size(columns=120, lines=30)
>>>

### In Thonny 4.1.6 shell :
Python 3.13.0 (C:\Program Files\Python\python.exe)
>>> import os
>>> print(os.get_terminal_size())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [WinError 6] Descripteur non valide
>>>

aivarannamaa, 2025-01-19:
Thonny's shell is not a proper terminal editor, that's why this operation is not supported. As a work-around, use Run => Run current script in terminal.

closed 2025-01-19 by aivarannamaa

MdV-dev, 2025-01-20:
Thank you for the explanation and circumvention :)