thonny/thonny #3442: [feature request] Upload directory

0wwafa, 2025-01-19:
Please allow a feature to upload a directory (or a zip file) to the root of the esp32.

Like:

/lib
 test1.mpy
test2.py
/
boot.py
code.py

etc etc

aivarannamaa, 2025-01-19:
This is already possible -- just open the target directory on the lower part of the file-browser, right-click on the source directory and select "Upload to ...":

[image]

0wwafa, 2025-01-19:
Sorry! I missed the file browser completely.
P.S.
Terminal imoplementation could be improved... I always end up disabling it and using putty...

0wwafa, 2025-01-19:
would be nice to have drag and drop in the remote folders view... easier when you have a messu harddrive :D

aivarannamaa, 2025-01-19:
Drag and drop from another app is not easy with the GUI toolkit Thonny is using.

closed 2025-01-19 by aivarannamaa