thonny/thonny #3446: thonny circuitpython usb port dropout problem

KarlD1239, 2025-01-21:
I'm using Thonny to program some Adafruit M0 express boards with circuitpython.  Everything works except that sometimes the board gets disconnected from the usb port without an apparent cause.   The feedback says that there is a backend error in that it can't find the port.  What do I need to do to correct this?

Running Windows 11 on a MS Surface Pro 7  
Thonny 4.1.7  and Python 3.10.11

I am running Adafruit CircuitPython 9.2.3 on 2025-01-17; Adafruit Feather M0 RFM69 with samd21g18
in this particular case. and have configured the interpreter for circuitpython generic

aivarannamaa, 2025-01-22:
When you receive this error again, select Tools => Open Thonny data folder and send me backend.log

Make sure you copy the file just after the error happens - when you reconnect or restart in Thonny, the file gets overwritten.

aivarannamaa, 2025-01-22:
Similar error was in #2585 and #2710 but I haven't seen it myself and I don't know what may cause this.