thonny/thonny #3453: "Upload to /" seems to have inconsistent behavior for ESP32

rosesonfire, 2025-01-27:
Sometimes when uploading a folder from computer to '/' of ESP32 using "Upload to /" the contents of the said folder are uploaded to '/' as well.

So for example, I want to upload a folder in computer 'My Folder' which has contents 'a.py', 'b.py' and 'c.py'.

```
My Folder
|--a.py
|--b.py
|--c.py
````

So I right click on 'My Folder' and click on "Upload to /". I expect to see the just the folder 'My Folder' uploaded to '/'.
```
My Folder
|--a.py
|--b.py
|--c.py
```

But I see instead 'a.py', 'b.py' and 'c.py' uploaded to root as well.

```
My Folder
|--a.py
|--b.py
|--c.py
a.py
b.py
c.py
```

This does not happen all the time. It happens once in a while. I have observed it in ESP32 and ESP32 S3

aivarannamaa, 2025-02-04:
Thank you for the report!

Here my thinking was that everything that gets selected for upload, ends up in the target folder, and that is what happened -- you selected "My Folder" and it ended up in `/`, together with its contents. You selected "a.py" and it ended up in `/`. Basically, you selected "a.py" twice -- via "My Folder" and directly.

I understand that this can be unintuitive. I just noticed macOS Finder acts differently -- when selecting a folder and a file in it and copying both of them to another location, the result was the same as just copying the folder. Same in Pycharm. In Windows Explorer I was not able to simultaneously select a folder and a file inside it.

I guess it makes sense to change this and simply ignore the selected items, which have their parent (or further ancestor) folder selected.

rosesonfire, 2025-02-04:
Thanks, such an modification will be much appreciated!