thonny/thonny #3460: py5 on Windows 10: Error occurred during initialization of VM

WLNDRMBK, 2025-02-05:
Sorry for a long post, but I've tried a lot now already, and it's always best to give all the details:

I am trying to run py5 sketches in thonny. But I can't get any py5 sketches to run. py5 is installed and active. The error I am getting:

```
Error occurred during initialization of VM
Could not find agent library instrument on the library path, with error: Can't find dependent libraries
Module java.instrument may be missing from runtime image.

Process ended with exit code 1.
```

[image]

I can run regular python scripts just fine in thonny. I installed Java for py5 specifically. I went with Java 17 and it's the only version of Java on my machine. Windows environment variables are set for this installation of Java. java -version and javac -version are giving me correct output in cmd, and I can also run java code from the command line.
But when I installed py5, thonny insisted on installing Java 17 again, but it did so in another place. This means I now have Java 17 installed in 2 different locations. I checked Thonny Options and there's a field for environment variable. 
I have tried setting thonny environment variable to the same location of my manual java installation: JAVA_HOME=C:\Program Files\Java\jdk-17
I have also tried setting both the thonny environment variable and the windows environment variable to point to the location of the jdk that thonny had installed automatically.
Both to no avail.

Any other things to try?

This is just a hunch but maybe there are some residual files from an earlier installation? I had tried getting py5 and thonny to run last year but gave up in frustration because of another java issue (I missed the fact it had to be Java 17 specifically). So I uninstalled everything then, but there might be some files left somewhere that are becoming a problem now. If that might be the case, how would I uninstall everything with no files remaining?

WLNDRMBK, 2025-02-05:
Update:
After a restart, I am now having the same problem I have had last year:

```
py5 is unable to start a Java 17 Virtual Machine.
This library requires Java 17 to be installed and a properly set JAVA_HOME environment variable.
Here is some debug info about your installation that might help you identify the source of this problem.
{'JAVA_HOME environment variable': 'C:\\Program Files\\Java\\jdk-17', 'jvm version': (0, 0, 0), 'default jvm path': 'C:\\Program Files\\Java\\jdk-17\\bin\\server\\jvm.dll'}
Traceback (most recent call last):
  File "C:\Users\DefUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\py5_tools\tools\run_sketch.py", line 72, in <module>
    main()
  File "C:\Users\DefUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\py5_tools\tools\run_sketch.py", line 63, in main
    imported.run_code(
  File "C:\Users\DefUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\py5_tools\imported.py", line 148, in run_code
    _run_code(
  File "C:\Users\DefUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\py5_tools\imported.py", line 304, in _run_code
    _run_sketch(sketch_path, classpath, exit_if_error)
  File "C:\Users\DefUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\py5_tools\imported.py", line 226, in _run_sketch
    import py5
  File "C:\Users\DefUser\AppData\Local\Programs\Python\Python313\Lib\site-packages\py5\__init__.py", line 75, in <module>
    raise RuntimeError("py5 is unable to start Java 17 Virtual Machine")
RuntimeError: py5 is unable to start Java 17 Virtual Machine

```

aivarannamaa, 2025-02-08:
You might get better advice from the authors of https://github.com/tabreturn/thonny-py5mode