thonny/thonny #3547: Keyboard Shortcuts Not Recognized When Using Hebrew Keyboard Layout

annonimono, 2025-02-18:
### **Keyboard Shortcuts Not Recognized When Using Hebrew Keyboard Layout**

**Description:**
When using the Hebrew keyboard layout in Thonny, keyboard shortcuts such as **Copy (Ctrl+C)**, **Cut (Ctrl+X)**, and **Paste (Ctrl+V)** are not recognized or do not work as expected.

**Steps to reproduce:**
1. Switch the keyboard layout to Hebrew.
2. Try using common shortcuts like **Ctrl+C**, **Ctrl+X**, and **Ctrl+V**.
3. Notice that these shortcuts are not functioning.

**Expected behavior:**
The **Copy**, **Cut**, and **Paste** shortcuts should work regardless of the active keyboard layout.

**Actual behavior:**
Shortcuts do not work when the Hebrew keyboard layout is active.

**Additional information:**
This issue occurs only when the Hebrew layout is in use. Switching to the default English layout restores the functionality of these shortcuts.

**Request:**
Please provide a list of all the available keyboard shortcuts for Thonny.

**Note:**
I am not very fluent in English, and I used GPT to help write this message.

aivarannamaa, 2025-02-23:
Thanks for the report! I'll investigate this later.

niknal357, 2025-12-05:
This is generally a wider issue (which affects users of non-QWERTY layouts). The correct fix is to read scancodes rather than keycodes for keybinds.