thonny/thonny #3555: Suggestion: open directory/project (support 'open in thonny' in Github Desktop)

adjamhub, 2025-02-24:
My idea is to allow directory opening just opening the file view there. Maybe the editor can open __init__ file, __main__ file or pyproject.toml.
All this just to better integrate Thonny in Github Desktop 'Open in' option.

If you are interested but short in time, I can try providing a patch about.

aivarannamaa, 2025-02-24:
Can you please rephrase your idea? I'm afraid I didn't get it.

adjamhub, 2025-02-25:
Sure!
Github Desktop ---> Open in external editor. Let Thonny support this option. 

Actually, if you launch "Thonny FILEPATH" you can open a file inside Thonny.
I'd like to add the option to launch "Thonny DIRPATH" and list the dir in the file view. We can add option also to open some files by default: init, main, pyproject.

aivarannamaa, 2025-02-25:
So you'd like to call Thonny with `thonny ~/my-projects/my-cool-project` from the command line? I think that's a very good idea! You are welcome to create a pull request for this. 

> We can add option also to open some files by default: init, main, pyproject.

I'm hesitant about this. I'd rather let the user to choose the file(s) to open each time.

adjamhub, 2025-02-25:
Very good then!
I'll try providing a pull request about