thonny/thonny #3557: Feature request: ability to open multiple files of the same name

FaramosCZ, 2025-02-25:
Hello,
I like Thonny very much ❤  , but I struggle when programming multiple ESP32 at once.

For example when I develop network communication, I need to quickly switch between the connected ESPs and work a bit here and there ... .

However Thonny does not allow for a file to be opened, while another file with the same name is already open.
Most basic example would be 'main.py' from multiple boards.

Could you please implement an ability to work with multiple same named files during one session ?

FaramosCZ, 2025-02-25:
I'm currently using
```
# flatpak info org.thonny.Thonny 

Thonny - Python IDE for beginners

          ID: org.thonny.Thonny
         Ref: app/org.thonny.Thonny/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 4.1.7
     License: MIT
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 79.7 MB
     Runtime: org.freedesktop.Sdk/x86_64/24.08
         Sdk: org.freedesktop.Sdk/x86_64/24.08

      Commit: d0110455c36afc1c620105de2ef7b8efc6fec1d64d22614058574d5f7bd56b5c
      Parent: d468307c86f8855b5dbd284c344732a152f1dd531b43083a49de58240fc6a3d6
     Subject: python3-setuptools_scm: Update setuptools_scm-8.1.0-py3-none-any.whl to 8.2.0 (26aae1e9)
        Date: 2025-02-22 02:02:57 +0000
```
on a Fedora host

aivarannamaa, 2025-02-26:
It's not only about the file names. Thonny manages non-local files via the active interpreter and it is designed to have single active interpreter at a time.

Thonny 5.0 will make it easier to launch several instances at a time, so you can open another instance for another device.

closed 2025-02-26 by aivarannamaa

FaramosCZ, 2025-02-26:
> Thonny 5.0 will make it easier to launch several instances at a time, so you can open another instance for another device.

Sounds great!

I will actually like several instances approach more, as it will help me keep the open files tab a bit less messy on bigger projects :)