thonny/thonny #3558: Python executable drop down does not display existing virtual environment

znoroozi, 2025-02-27:
I created a virtual environment by clicking "New Virtual Environment" under "Interpreter Tab. The virtual environment was created and it path was added to the Python executable list. When I closed Thonny and re-opened it, the path was gone so I could not select it. I can only activate the virtual environment from Terminal. Please help.

aivarannamaa, 2025-03-01:
Thank you for the report! I'll make sure that Thonny 5.0 will not have this problem.

WhitespaceQ, 2025-03-06:
I cannot reproduce this bug with https://github.com/thonny/thonny/releases/tag/v4.1.7

Can you share your environment detail?

eeshorts20, 2026-03-17:
I think the issue is the pull-down only allows seven items.