thonny/thonny #3560: Problemas al abrir Thonny

JeanPRosales, 2025-03-01:
Buenos días. He instalado varias veces Thonny para windows, y siempre me sale el mismo error nada más abrir Thonny. El error es este:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\cpython_frontend\cp_front.py", line 39, in __init__
    self._send_msg(ToplevelCommand("get_environment_info"))
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 1157, in _send_msg
    self._proc.stdin.write(serialize_message(msg) + "\n")
OSError: [Errno 22] Invalid argument

Process ended with exit code 9009.
No sé qué es lo que puede ocurrir. Muchas gracias por la ayuda.

aivarannamaa, 2025-03-01:
Which Windows version are you using?

JeanPRosales, 2025-03-02:
Hola! Tengo instalado windows 10 pro,

aivarannamaa, 2025-03-02:
Is it 32- or 64-bit Windows? Can you post a screenshot of Thonny's About box (Help => About) ?

JeanPRosales, 2025-03-03:
Hola! Es windows 64. Me instalé los 2 exe que me pusiste, y me sigue dando el mismo problema. Te dejo una captura de pantalla del problema:

[image]

aivarannamaa, 2025-03-21:
If you have 64-bit Windows, then you need https://github.com/thonny/thonny/releases/download/v4.1.7/thonny-4.1.7.exe

You could try this:

1. Launch Thonny
2. Select "Tools => Open Thonny data folder"
3. Close Thonny
4. Delete everything in the data folder
5. Reopen Thonny