thonny/thonny #3561: Install micropython on fresh Pidco2, from Thony running on Ubuntu24.04

gaddih1, 2025-03-02:
I uinstalled Thony for Pico2 from Snap on Ubuntu 24.04. I pressed the BOOTSEL button on the Pico2 when connecting the USB untill I heard a beep.  I can see the content on the mass stoarge RS2350, hence, I know I am connected to the Pico.  However when I try to install Micropython  (clicking the bottom right corner of the Thonny window) , the "target volume" line does not populate.  Any additional action results results in the  following error: 

Traceback (most recent call last):
  File "/snap/thonny/239/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/snap/thonny/239/lib/python3.10/tkinter/__init__.py", line 839, in callit
    func(*args)
  File "/snap/thonny/239/lib/python3.10/site-packages/thonny/workdlg.py", line 216, in _keep_updating_ui
    self.update_ui()
  File "/snap/thonny/239/lib/python3.10/site-packages/thonny/plugins/micropython/base_flashing_dialog.py", line 151, in update_ui
    targets = self.find_targets()
  File "/snap/thonny/239/lib/python3.10/site-packages/thonny/plugins/micropython/uf2dialog.py", line 34, in find_targets
    for vol in list_volumes(skip_letters=["A"])
  File "/snap/thonny/239/lib/python3.10/site-packages/thonny/misc_utils.py", line 112, in list_volumes
    return list_volumes_sync()
  File "/snap/thonny/239/lib/python3.10/site-packages/thonny/udisks.py", line 19, in list_volumes_sync
    return loop.run_until_complete(list_volumes())
  File "/snap/thonny/239/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/snap/thonny/239/lib/python3.10/site-packages/thonny/udisks.py", line 44, in list_volumes
    block_devices = await interface.call_get_block_devices({})
  File "/snap/thonny/239/lib/python3.10/site-packages/dbus_next/aio/proxy_object.py", line 79, in method_fn
    msg = await self.bus.call(
  File "/snap/thonny/239/lib/python3.10/site-packages/dbus_next/aio/message_bus.py", line 305, in call
    await future
  File "/snap/thonny/239/lib/python3.10/site-packages/dbus_next/aio/message_bus.py", line 365, in _message_reader
    if self._unmarshaller.unmarshall():
  File "/snap/thonny/239/lib/python3.10/site-packages/dbus_next/_private/unmarshaller.py", line 304, in unmarshall
    self._unmarshall()
  File "/snap/thonny/239/lib/python3.10/site-packages/dbus_next/_private/unmarshaller.py", line 243, in _unmarshall
    self.read(16, prefetch=True)
  File "/snap/thonny/239/lib/python3.10/site-packages/dbus_next/_private/unmarshaller.py", line 91, in read
    raise EOFError()
EOFError

Can anyone please help?

aivarannamaa, 2025-03-02:
I'm linking another open issue, which mentions a crash inside `dbus_next`: #3427. These may or may not be related.

Unfortunately I'm not familiar with this technology myself. I'm considering making `dbus_next` as a fallback to `mount` command (currently it is vice versa).

gaddih1, 2025-03-02:
Update: I installed Thonny from a debian  with "apt install Thotnny".   This version did recognize the Pico2 board, and even seemed to install Micro-Python (Raspberry Pi Pico) on the Pico2.   However, the installation on the Pico never actually occured.