thonny/thonny #3566: "Step Into" in debugger, display area is not long enough for multibyte string

kubomamoru, 2025-03-05:
When using the small step operation "Step Into" in the debugger, the display area is not long enough for multibyte strings.
Could you solve this problem by using unicodedata.east_asian_width?
[https://docs.python.org/3/library/unicodedata.html#unicodedata.east_asian_width](https://docs.python.org/3/library/unicodedata.html#unicodedata.east_asian_width)

[image]

WhitespaceQ, 2025-03-06:
Can you provide a minimal reproducible example?

kubomamoru, 2025-03-06:
Thank you for your reply.
The sample code is as follows:
[example_len_multibyte_py.txt](https://github.com/user-attachments/files/19100254/example_len_multibyte_py.txt)

WhitespaceQ, 2025-03-06:
Thank you for the example. I can fix the behavior to match your expected result, however, I’m not entirely confident in my implementation to submit a pull request just yet. Specifically, the expression `sum([(1, 2)[east_asian_width(c) in 'FWA'] for c in string])` is giving me an off-by-one result, requiring me to subtract 1 from the final calculation.

Admittedly, my experience with multi-locale applications is limited, so I’m still exploring a better solution to address this accurately. If you’d like, I can provide a diff so you can review and apply the patch yourself.

[image]

kubomamoru, 2025-03-07:
Thank you for your excellent solution so quickly.
I only use English and Japan, and have no experience of other locales.
If possible, please provide the diff code.

WhitespaceQ, 2025-03-07:
https://github.com/thonny/thonny/compare/master...WhitespaceQ:thonny:fix-multibyte-in-debugger

kubomamoru, 2025-03-10:
Thank you for providing the fix code.
My fix and results are as follows:

[image]

[image]

[image]

aivarannamaa, 2025-05-04:
@WhitespaceQ, thanks for the suggestion! I'll investigate it further and probably include it later.