thonny/thonny #3572: BUG: Thonny portable on Win10 needs "paramiko" for SSH2 but fails to install

shulist, 2025-03-15:
I recently installed **Thonny 4.1.7 Portable** on Win10. When setting up access to a development server using SSH Thonny throws a message `"This back-end requires and extra package named "paramiko"....` So following the provided instructions I do the following:
1) [Tools], [Manage plug-ins]
2) enter "paramiko"
3) its the first plug-in listed.
4) I select it and then everything hangs.

Can anyone shine a light on this issue and perhaps provide a solution?

shulist, 2025-03-16:
So I experimented with installing another plug-in -- Thonny-autossave. Interestingly the plug-in was found, however,  when selected it seemed to attempt to install, but moments later I received the following dump.

Traceback (most recent call last):
  File "C:\PortableApps\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 551, in poll_fetch_complete
    self._show_package_info(name, data)
  File "C:\PortableApps\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 569, in _show_package_info
    self._set_state("idle")
  File "C:\PortableApps\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 287, in _set_state
    widget["state"] = tk.NORMAL
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 1657, in __setitem__
    self.configure({key: value})
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 1646, in configure
    return self._configure('configure', cnf, kw)
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 1636, in _configure
    self.tk.call(_flatten((self._w, cmd)) + self._options(cnf))
_tkinter.TclError: invalid command name ".!pluginspipdialog.!frame.!panedwindow.!frame2.!frame.!button"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 814, in callit
    func(*args)
  File "C:\PortableApps\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 557, in poll_fetch_complete
    self._show_package_info(
  File "C:\PortableApps\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 569, in _show_package_info
    self._set_state("idle")
  File "C:\PortableApps\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 287, in _set_state
    widget["state"] = tk.NORMAL
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 1657, in __setitem__
    self.configure({key: value})
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 1646, in configure
    return self._configure('configure', cnf, kw)
  File "C:\PortableApps\Thonny\lib\tkinter\__init__.py", line 1636, in _configure
    self.tk.call(_flatten((self._w, cmd)) + self._options(cnf))
_tkinter.TclError: invalid command name ".!pluginspipdialog.!frame.!panedwindow.!frame2.!frame.!button"

shulist, 2025-03-16:
I think I've figured this out -- maybe. As I am using  a 32 bit OS on the development server I perhaps wrongly assumed I needed to run the 32 bit version of Thonny Portable ( thonny-py38-4.1.7-windows-portable.zip ) as a further experiment I installed the 64 bit version (
thonny-4.1.7-windows-portable.zip ) and lo and behold the plug-in is already installed in that version. 
So I think it may be a problem with the 32 bit version of Thonny?

aivarannamaa, 2025-05-04:
Thank you for the report and sorry for the late reply!

Thonny 5.0 will have refactored plugin management, so most likely this issue won't happen anymore (or at least in the same form)

closed 2025-05-04 by aivarannamaa