thonny/thonny #3580: Size of autocompletion box

mcaipoupoule, 2025-03-28:
The autocompletion box is too small on my configuration.
Is there a (manual?) way to change its size while waiting for a way to do so to be coded from the user interface or in the settings?

Thanks in advance !

{
Linux Mint 22.1 Cinnamon
Cinnamon 6.4.8
kernel 6.8.0-56-generic
X11
Thonny 4.1.7
}

aivarannamaa, 2025-05-04:
Thanks for the report!

No, the box doesn't have means for manual resizing. Can you post a screenshot of your problem?

mcaipoupoule, 2025-05-10:
One example among many :

The last two elements are indistinguishable.
This is even truer with a list of constants.

[image]

In an ideal world, you could place the mouse cursor at the bottom right and resize.  **;-)**