thonny/thonny #3583: Grammar fix in Support Ukraine wiki page

Pryodon, 2025-03-30:
In the sentence highlighted in bold at the bottom of the first section, "It looks seems Trump's goal is to put Ukraine into weakest possible position before the peace negotiations." The word "looks" needs to be removed. Or, you could say "It looks as if Trump's" and remove the word "seems." This is how I would write the sentence... Notice that I added the word "the" before the word "weakest."

"It seems Trump's goal is to put Ukraine into the weakest possible position before the peace negotiations."

The page I am referring to is this [link](https://github.com/thonny/thonny/wiki/Support-Ukraine):

I would do a pull request to fix the issue, but I really have no idea how to even begin doing that. I am still trying to figure out how to use git.

qiuqiu035, 2025-04-23:
Hi! I'd like to fix this issue. I can submit a pull request with the correct sentence. 😊

aivarannamaa, 2025-04-24:
Thank you for the notification and sorry for the late reply!

closed 2025-04-24 by aivarannamaa