thonny/thonny #3584: [4.1.7] Crash when click on install from local file

lobsterCARAMEL, 2025-03-31:
mac os, tried on intell and M1. crash happens each time. thonny is currently unusable. 

reproduce: go to Tools - Manage packages - Install from local file. Boom. 

 version [4.1.7]

ChristopherPAndrews, 2025-03-31:
It looks like this is a duplicate of earlier reports:

https://github.com/thonny/thonny/issues/3372
https://github.com/thonny/thonny/issues/3359

aivarannamaa, 2025-05-04:
Thank you for the report!

Thonny 5.0 will have reworked package management. I'll test this case later.

aivarannamaa, 2025-05-31:
Fixed in 0060404261b67fca479b0b420dd9762411777ffd

closed 2025-05-31 by aivarannamaa

danschlet, 2025-12-26:
I can't find a (pre-release) version of 5.0. Is there any workaround, so we Mac users can use Thonny to install local files or do we have to wait for the 5.0 final release? Thank you for any answer.