thonny/thonny #3586: Thonny does not start under Raspian

GuentherGredy, 2025-04-02:
Hello,
I'm new here and a beginner on Raspberry Pi (3B+ with Raspian, configured as headless). I read that Thonny comes with the Raspian OS and wanted to start it
1st: on Pi directly from my home directory simply entering "thonny"
2nd: on Windows PowerShell with ssh user@ipaddress -X thonny.
Both trials lead to the same result - a lot of messages I cannot understand and therefor don't know how to solve the issue.
I did two screenshots, First.PNG and Second.PNG, they depict the complete set of messages in that order.

[image]

[image]

I kindly ask you to help me out of this incomprehensible situation. What is my mistake?
Thanks in advance and BR
Guenther

BertLindeman, 2025-04-02:
Hi Guenther,

The short answer:
You run a headless server, so the graphical environment that thonny needs is not available.

If you really need thonny on the headless raspbian server, you need to install and configure a graphical environment.
I am not well based in that tools, so I can not help you there.

Bert

GuentherGredy, 2025-04-03:
Hello Bert,

thank you very much for your quick reply.

I apparently misunderstood a post online that was about a "headless" Raspberry Pi with Thonny.
Actually, what you wrote makes perfect sense! Since I've already configured my Pi well, the question
arises whether the graphical environment can be installed later. Do you know anything about it?

Best regards, Guenther

P.S.: We already know each other from OsmAnd; you program there, right? Were do you live?

closed 2025-05-04 by aivarannamaa