thonny/thonny #3631: Is there a way to create a non-political version for use in the classroom?

Bc-gar, 2025-05-28:
Hi there!
First of all: Great job!
Personally, I stand with Ukraine.
The problem is that teachers in Germany are not allowed to "enforce" political opinions on students. 
So as long as the Ukrainian flag is on the screen, we cannot use Thonny in the classroom. But I'd really love to do that as it seems to me an ideal IDE for beginners. 
So is there are way to hide the flag?
Thanks in advance!
Christiane

kolotxoz, 2025-05-28:
For the moment, yes, there is.

You have to modify the workbench.py file, in windows is usually stored in this folder:

C:\Program Files (x86)\Thonny\Lib\site-packages\thonny\workbench.py

An you have to comment this block of code:

 self.add_command(
            "SupportUkraine",
            "help",
            tr("Support Ukraine"),
            self._support_ukraine,
            image="Ukraine",
            caption=tr("Support"),
            include_in_toolbar=True,
            group=101,
        )

It should look like this:

 #self.add_command(
        #    "SupportUkraine",
        #    "help",
        #    tr("Support Ukraine"),
        #    self._support_ukraine,
        #    image="Ukraine",
        #    caption=tr("Support"),
        #    include_in_toolbar=True,
        #    group=101,
        #)

[image]

Bc-gar, 2025-05-29:
Thank you very much! I‘ll try this straightaway!