thonny/thonny #3637: "Error when initializing backend" after fresh install.

caisstuff, 2025-06-08:
Howdy, the following occurs on Windows 11 after installing the latest version of Thonny (4.1.7).

> Traceback (most recent call last):
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 530, in _start_runner
>     self._runner.start()
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 155, in start
>     self.restart_backend(False, True)
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 731, in restart_backend
>     self._proxy = backend_class(clean)
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 93, in __init__
>     super().__init__(clean)
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 32, in __init__
>     super().__init__(clean, running.get_front_interpreter_for_subprocess())
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 999, in __init__
>     self._start_background_process(clean=clean)
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_front.py", line 137, in _start_background_process
>     super()._start_background_process(clean=clean, extra_args=extra_args)
>   File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 1093, in _start_background_process
>     self._proc = subprocess.Popen(
>   File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 971, in __init__
>     self._execute_child(args, executable, preexec_fn, close_fds,
>   File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 1456, in _execute_child
>     hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
> OSError: [WinError 87] The parameter is incorrect

The IDE does not work at all and I'm not sure if this is a user issue, some help and info would be appreciated.

an-aegis, 2025-08-03:
ensure your python is actually where its looking

[image]
(go into run at the top and click configure interpreter, the options window will appear and your screen should look functionally identical to mine if everything's good)

caisstuff, 2025-08-08:
Howdy! Mines located directly in programs, and also this started working suddenly within the last month though sometimes it will throw the same error, it might be a me issue after all.

[image]

Thanks for responding anyways.

closed 2025-08-08 by caisstuff