thonny/thonny #3655: Installing packages using tools is broken on Linux

wwwqr-000, 2025-07-13:
Installing packages using tools is broken on Linux. The new Python versions require you to use a virtual environment. When the interpreter is changed to a virtual Python environment, it only has effect on the current shell or terminal. It doesn't get used when installing packages using "Tools"

villares, 2025-08-07:
Hi @wwwqr-000 I'm on Linux and I didn't have this problem, so let me see if I can understand what is going on and help! How did you install Thonny?

I have installed Thonny using this (from https://thonny.org) :
`bash <(wget -O - https://thonny.org/installer-for-linux)`

robertlq, 2025-09-01:
Hi, I'm facing the same problem with Thonny for Pico Micropython development: No install / uninstall of packages is possible, a hundred of lines of error text without any meaning.
My only work-around is to use an old Ubuntu PC with python 3.10 for package installation, there the package management is working as usual.
Thanks a lot for any help!

villares, 2025-09-01:
> Hi, I'm facing the same problem with Thonny for Pico Micropython development: No install / uninstall of packages is possible, a hundred of lines of error text without any meaning. My only [...]

Hi @robertlq !

When Thonny uses the Micropython as the interpreter (is it for using on a Raspberry Pi Pico?) the packages you can install are very limited, because it is supposed to run on a small microcontroller board.

From which OS are you launching Thonny and having this issue? Can you changing the interpreter at the Options panel?

I know it is possible to install py5 on a "full" ARM/Raspberry Pi, with some work...

solmoller, 2026-02-15:
Same problem here on a spanking new Raspberry Pi OS, virtual environment not recognized.

villares, 2026-02-15:
> Same problem here on a spanking new Raspberry Pi OS, virtual environment not recognized.

Hi @solmoller, have you tried installing Thonny with: `bash <(wget -O - https://thonny.org/installer-for-linux)` or installed Thonny as a Python package?

Another thing to try is asking Thonny to create a new Vitual Env in the Options > Intepreter panel...

villares, 2026-02-15:
> Thonny should be set up so there is a default virtual environment, or maybe a better error message when trying to install plugins before such environment exists

This sounds reasonable. An the Thonny maintainer is always looking for improvements. Mind you that I'm just a user like you, not the maintainer.

As a workaround, I would suggest you could try this... 

- Open the terminal and create an env yourself with `python -m venv thonny-env` (or another env name you preffer)
- Then `source thonny-env/bin/activate`
- So that now you can install Thonny as module with `pip install thonny`
- And then you can launch it with `python -m thonny`.

Maybe this could help...

aivarannamaa, 2026-04-26:
Please check, whether the latest release (https://github.com/thonny/thonny/releases/tag/v5.0.0) has made anything better.

wwwqr-000, 2026-05-24:
Sorry for my late reply. This fixed the problem for me, thank you so much <3

closed 2026-05-24 by wwwqr-000