thonny/thonny #3672: Unable to install thonny in windows 11!

app-src, 2025-08-24:
I had thonny installed on my PC
By mistake someone removed the install folder
After that I have rearranged he Drives so now the drive where thonny was earlier does not exist on PC.
So now if I try reinstalling thonny some how it is detecting the previously installed location and only trying to upgrade the already existing thonny in previously installed location(which does not exits on the PC).

Deleted all the thonny related files but still somehow it is detecting the previously installed location and not giving me an option to update it while reinstalling.

aivarannamaa, 2025-08-25:
Did you uninstall Thonny or simply removed the folders? Uninstall is supposed to remove knowledge about previous location.

app-src, 2025-08-25:
I didn’t uninstall it, but this situation isn’t good. There should be a way to fix it

aivarannamaa, 2025-08-25:
Can you uninstall it now?

app-src, 2025-08-25:
Nope. Deleted the uninstaller also :⁠'(

aivarannamaa, 2025-08-25:
In this case you need to open regedit, search for entries mentioning Thonny and delete these (or at least the ones mentioning its installation path)

app-src, 2025-08-25:
I was being impatient, Ctrl + F does the job 👍🏼

closed 2025-08-25 by app-src